majekw/optiboot #1: Write to flash doesn't work with the ATmega644p

MCUdude, 2016-06-07:
Hi! I'm trying to figure out why this bootloader won't write to the ATmega644's flash memory. It works fine with the ATmega1284. I've tried the bootloader in the master and the supermaster branch, but no luck. 
This is the make command I'm using:
`make atmega644p AVR_FREQ=16000000L BAUD_RATE=115200 UART=0 LED=B0 LED_START_FLASHES=2 -BIGBOOT`

I've tried yours and my own example code. The pre stored characters in the array seems untouched.. Do you have any idea why this isn't working. The ATmega644p and 1284p are quite similar

majekw, 2016-06-07:
I bet it's issue with entry address. ATmega644 doesn't have RAMPZ but it's BIGBOOT as smallest bootloader size is 1024B for this chip.
Please try to replace 511 with 1023 at https://github.com/majekw/optiboot/blob/supermaster/optiboot/examples/flash_program/optiboot.h#L85 and check if this hack will help. If yes, I need to rewrite this part a little.

MCUdude, 2016-06-07:
Thanks! That did the trick. I tried to use the preprocessor functionality, but this didn't trigged the correct line..
`#ifdef RAMPZ
  typedef uint32_t optiboot_addr_t;
  const do_spm_t do_spm = (do_spm_t)((FLASHEND-1023+2)>>1);
#else
  typedef uint16_t optiboot_addr_t;
  #if defined(__AVR_ATmega644__) || defined(__AVR_ATmega644p__)
  	const do_spm_t do_spm = (do_spm_t)((FLASHEND-1023+2)>>1);
  #else
  	const do_spm_t do_spm = (do_spm_t)((FLASHEND-511+2)>>1);
  #endif	
#endif`

Where does *__AVR_ATmega644p__* and *__AVR_ATmega644__* get defined?

majekw, 2016-06-07:
I think that this will be better (support also Atmega64):
```
#ifdef RAMPZ
  typedef uint32_t optiboot_addr_t;
#else
  typedef uint16_t optiboot_addr_t;
#endif

#if FLASHEND > 65534
  const do_spm_t do_spm = (do_spm_t)((FLASHEND-1023+2)>>1);
#else
  const do_spm_t do_spm = (do_spm_t)((FLASHEND-511+2)>>1);
#endif
```

__AVR_ATmega... are defined by compiler when you choose target chip.

MCUdude, 2016-06-07:
Great! That did the trick! 😉 
I'm almost finished crapping together some higher level fiction to read/write to flash. I decided to store the functions in the optiboot.h file, since it doesn't need to be object oriented anyway. I also put together an Arduino sketch where one can use the serial monitor to read/write and browse the memory:
[Read_and_write.zip](https://github.com/majekw/optiboot/files/303732/Read_and_write.zip)

majekw, 2016-06-10:
Thank you @MCUdud, I updated spm and supermaster branches with this fix.

closed 2016-06-10 by majekw