majekw/optiboot #2: avrdude: verification error

DzurDzen, 2016-06-09:
Hi. When I try to upload a sketch provides verification error.
avrdude: verification error, first mismatch at byte 0x0000
         0x06 != 0x07
avrdude: verification error; content mismatch

Bootloader compiled with Arduino IDE 1.5.4 and with 1.0.5 (hex are identical). I tried avrdude versions of 6.1-6.3. When flashing back to original bootloader (with usbasp), the problem disappears.

Seeeduino Mega (atmega2560) from seeedstudio.com
Windows 7
Arduino IDE 1.6.9
m2560 branch

majekw, 2016-06-10:
Strange. I tried today with Arduino 1.6.9 and Avrdude 6.3 and it worked fine (on Linux).
Could you check this ready hex with bootloader: https://raw.githubusercontent.com/majekw/optiboot/supermaster/optiboot/bootloaders/optiboot/optiboot_atmega2560.hex ?
When you replaced Avrdude on Arduino, did you overwrote avrdude or did also something else?
I'll try to check this also on Windows later.

DzurDzen, 2016-06-11:
I checked ready hex from supermaster branch. Everything is fine. When compiling from a m2650 branch hex a little smaller in size. Now sketches uploaded via USB and via esp-link (firmware for esp8266).
Thanks

majekw, 2016-07-03:
@DzurDzen, could you provide .lst file for this non-working bootloader you compiled from m2560 branch? I'm curious what wrong could be with it as it should work.

majekw, 2016-07-03:
Same issue here: http://forum.arduino.cc/index.php?topic=410068.0

crgor, 2016-07-05:
Majek, Unfortunately I couldn't attach the lst file over there at Arduino Forum. Attached is the LST(renamed to txt) file for the optiboot supermaster branch compiled with Arduino IDE 0023 for atmega 2560.

[optiboot_atmega2560.txt](https://github.com/majekw/optiboot/files/347045/optiboot_atmega2560.txt)

DzurDzen, 2016-07-05:
Hex and lst

[optiboot_atmega2560.hex.txt](https://github.com/majekw/optiboot/files/347052/optiboot_atmega2560.hex.txt)
[optiboot_atmega2560.lst.txt](https://github.com/majekw/optiboot/files/347053/optiboot_atmega2560.lst.txt)

majekw, 2016-07-21:
Thank you @DzurDzen and @crgor!
I looked into these lst, but besides compiler generated code differences I can't see anything suspicious. I'll try to burn it into my Atmega2560 in following days and look what will happen.

majekw, 2016-10-20:
This should fix this issue: https://github.com/majekw/optiboot/commit/bd0a623dfead7d60fb9766519b6891a7df224ccd

Problem was that older versions of avr-gcc shipped with Arduino earlier than 1.6.x doesn't define RAMSTART for Atmega2560, so Optiboot incorrectly assumes it's 0x100. At the beginning of memory is buffer, so some i/o registers are destroyed during upload and that leads also to flash corruption.
For me everything worked fine because I used always 1.6.x to compile things and that's why precompiled hex in supermaster is correct and works.
Thank you all for commitment, and @matthijskooijman for finding this bug.

commit 9e33509d (2017-05-17): documentation fix #2
closed 2016-10-20 by majekw