majekw/optiboot #3: Create some kind of flash data dumper

MCUdude, 2016-08-12:
Hi!
As you may know I've been using you Optiboot fork with my [Arduino cores](https://github.com/MCUdude) for the last couple of months. I also created a [flash read/write example](https://github.com/MCUdude/MightyCore/blob/master/avr/libraries/Optiboot_flasher/examples/SerialReadWrite/SerialReadWrite.ino) that lets you read/write to any page in the flash memory. It works great, but one thing I've missed is the ability to dump the content of the flash memory. Lets say you've been logging some data to the flash memory, and want to analyze this on you computer. This is hard, especially because the flash content gets erased after every upload.

First, is it possible to prevent the whole flash content from being erased after every code upload? Is there an elegant way to include some "flash dumping code" in your sketch that's easy to trigger (e.g double tap the reset button, then dump the content on the serial monitor)? If this feature can be added this flash writer feature could really show its strengths!

MCUdude, 2016-08-16:
Dumping the flash content didn't turn out to be so hard. A simple for loop did the trick:

``` c++
     for(uint16_t page = 1; page < NUMBER_OF_PAGES+1; page++)
      {
        Serial.print(F("Page "));
        Serial.print(page);
        Serial.print(F(": "));
        optiboot_readPage(flashSpace, ramBuffer, page, blankChar);
        Serial.println((char*)ramBuffer);
      }
```

The "hard" part is to implement this in the code. I was thinking about creating a data logging example where the data can easily be dumped. These are options I've been thinking about:
* Send some kind of serial command
* Add a physical button that trig the "dump code"
* Always dump data on reset
* Make use of [double-reset](https://github.com/Optiboot/optiboot/issues/181)

I'm no bootloader guru; Would it be possible to add some kind of double reset feature without too much effort?

majekw, 2016-10-20:
Hello Hans, sorry for late answer. I appreciate what you are doing and I hope that some day I put your changes back also to my fork. I really like moving hex files to another directory and support for other chips.
But back to the case.

1. Bootloader should do only as small programmer inside chip, not anything more. And it should be as little invasive as possible.
It may expose some extra features as flash programming, but without bootloader it's also possible by putting spm in right memory region. Even worse (from application perspective), bootloader disables spm possibility in user program, so workaround should be somehow implemented.

2. Application writes to flash and knows memory locations where data is stored, so application should also read from it. It's also application job to implement suitable protocol to dump data.

3. Using Avrdude you can read all flash :-) And only the whole flash at once :-(
Just try to run Avrdude with the same parameters as Arduino calls it during write, but change last part from `-Uflash:w:some.hex:i` to `-Uflash:r:somefile.dump:r` and you'll get binary/raw dump of all flash.
Then just extract interesting part.

4. The way Optiboot works, it does nothing during 'chip erase'. It erases only pages currently written, so no other pages are destroyed.
It's perfectly doable to assign some memory to program (with some unused space reserved for future program growth) and the rest for data (above some address). Then you could upload new code as long as it fits in it's part without touching data.

So, it's no need to modify anything in Optiboot to achieve your goal :-)

closed 2017-09-03 by majekw