majekw/optiboot #4: Optiboot on ATmega2561

MCUdude, 2017-02-03:
Hi!
First, thank for for this AWESOME fork of Optiboot! As you might have seen, I'm using it for all my "relevant" Arduino cores, and they are getting quite popular 😃 

I was able to get a few samples of the ATmega1281 and ATmega2561, and I'm thinking about adding Arduino support for these through [MegaCore](https://github.com/MCUdude/MegaCore). There was no problem adding the ATmega1281 to the makefile, and compiling (your fork of) Optiboot for it. However, I'm a little more uneasy about the ATmega2561. It's similar to the ATmega2560, which you've added support for. As specified in several Arduino forum topics, the "original" stk500 protocol that Optiboot is based on doesn't support devices with more than 128kB of flash, while the ATmega2560/2561 has 256kB. In [this thread](https://forum.arduino.cc/index.php?topic=410068.0) westfw points this out, and [this pull request](https://github.com/Optiboot/optiboot/pull/190) has not been added to the Optiboot master.

### Questions:

* Why isn't your "little fix" accepted by the Optiboot maintainer?
* Is this "fix" considered safe, as in 100% stable even if the code occupies more than 128kB of flash?
* Is there a reason why it shouldn't work on the ATmega2561?

Thanks!

majekw, 2017-02-03:
Hello Hans!

> Is this "fix" considered safe, as in 100% stable even if the code occupies more than 128kB of flash?

Works for me:-) and according to comments on forum orgithub issues also for some other people.
I would be much more worried about avr-gcc which is broken by design and have some hacks to support more flash, although it also works:-) 

> Is there a reason why it shouldn't work on the ATmega2561?

In general: no.
But it wouldn't work on very old chips, both Atmega2560 and Atmega2561 with chip revision A had a problem with flash writing (right now these should be manufactured at revision F), see errata in datasheet.

Just give it a try, I think it should work without problems.

MCUdude, 2017-02-04:
Great! I'll solder the chips tomorrow and report back 😊

MCUdude, 2017-02-04:
The bootloader works perfect in the ATmega1281 and ATmega2561! 😄  However, I'm not able to write to flash using my [SerialReadWrite.ino](https://github.com/MCUdude/MegaCore/blob/master/avr/libraries/Optiboot_flasher/examples/SerialReadWrite/SerialReadWrite.ino) sketch. After I've selected what page I'm going to write to, and written the text I want to store, I hit the termination character, which is the **@** character. Instead if writing to flash the program resets, and we're back to start without any new content in the flash memory I've used your m2560 branch. Doesn't this supports flash write?

majekw, 2017-02-04:
m2560 branch have only patches for 256KB ram devices. It exists mostly for original Optiboot pull request.
Please use supermaster branch as it have all features and patches.

MCUdude, 2017-02-04:
Thanks, that did the trick! I'll release the files soon 👍

closed 2017-02-04 by MCUdude