majekw/optiboot #5: Optiboot build for mega2560 doesnt seem to work

milkpirate, 2017-04-10:
Hey,

I am trying to get the optiboot bootloader to work on my MKS Gen (it is a board for 3D printing - RAMPS compatible - Arduino Mega compatible). This board features a ATmega2560 @ 16MHz. I would like to use 1Mbaud for uploads and so I tried to build it myself with that bausrate:

```
F_CPU=16000000L AVR_FREQ=16000000L BAUD_RATE=1000000 BIGBOOT=1 make atmega2560
```

The resuting hex file flashes the LED (D13... standard) really fast 3 times approximatly every second. When I try to upload some code, even at 115200 baud, with the Arduino IDE, it ru s into timeout errors:

```
C:\Users\USER\Documents\Blink>"C:\Program Files\arduino-1.8.0\hardware\tools\avr/bin/avrdude" -C"C:\Program Files\arduino-1.8.0\hardware\tools\avr/etc/avrdude.conf" -v -patmega2560 -cstk500v2 -PCOM24 -b1
15200 -D -Uflash:w:C:\Users\USER\AppData\Local\Temp\arduino_build_502770/Blink.ino.hex:i

avrdude: Version 6.3, compiled on Dec 16 2016 at 13:33:19
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files\arduino-1.8.0\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM24
         Using Programmer              : stk500v2
         Overriding Baud Rate          : 115200
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: Unknown
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
...
```

Could you help me out? Am I missing something and if so what?

MCUdude, 2017-04-10:
Try using `-cstk500`instead of `-cstk500v2` when uploading

milkpirate, 2017-04-11:
Same same... :(
```
C:\Users\USER\Documents\Blink>"C:\Program Files\arduino-1.8.0\hardware\tools\avr/bin/avrdude" -C"C:\Program Files\arduino-1.8.0\hardware\tools\avr/etc/avrdude.conf" -v -patmega2560 -cstk500 -PCOM24 -b1
1000000 -D -Uflash:w:C:\Users\USER\AppData\Local\Temp\arduino_build_502770/Blink.ino.hex:i

avrdude: Version 6.3, compiled on Dec 16 2016 at 13:33:19
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files\arduino-1.8.0\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM24
         Using Programmer              : stk500
         Overriding Baud Rate          : 1000000
avrdude: serial_baud_lookup(): Using non-standard baud rate: 1000000avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500
         Programmer Model: Unknown
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
...
```

milkpirate, 2017-04-11:
Ok pretty strange...  I changed the programmer type `-c` from `stk500`/ `stk500v2` to `arduino` (so `-carduino`) and now it worked. Can someone confirm that? Can I still access all then 256k?

majekw, 2017-04-11:
@milkpirate I'm glad to hear you made it. Does it really work well at 1Mbit?

@MCUdude, good catch with this stk500v2.

Arduino uses -carduino as it's a little different programmer in avrdude, main difference is on making reset pulse on DTR/RTS at start, and different way of reading chip signature.

MCUdude, 2017-04-11:
> Arduino uses -carduino as it's a little different programmer in avrdude, main difference is on making reset pulse on DTR/RTS at start, and different way of reading chip signature.

so it's the -carduino option that makes sure the device signature is retrieved from the bootloader, and not the AVR hardware itself?

majekw, 2017-04-11:
No, it's just different command send to programmer.
Strange thing is that I couldn't find in avrdude sources that generic stk500 have any way of reading signatures (it's specified in protocol, but it's probably not implemented?).

In Optiboot device signature is compiled in bootloader, so for example if you put Atmega328PA hex into Atmega328PB, bootloader will advertise itself to avrdude as PA chip, not PB.
Reading signatures from chip at runtime is also possible, but it will add more code, so I think it's not worth it.

milkpirate, 2017-04-12:
@majekw: Yes. The harware supports 1Mbaud as long as you clock the AVR with an appropriate frequency. See http://www.gjlay.de/helferlein/avr-uart-rechner.html - error should be <3%. I guess the bootloader handles ACK's and NACK's if the bytes comming too fast from UART and stalls untill the flash is ready again.

closed 2017-09-03 by majekw