majekw/optiboot #8: is it possible to detect "flash write" feature?

uzi18, 2017-08-07:


majekw, 2017-08-08:
I don't think that that's 100% reliable method, but good guess would be:
1.Check for size of bootloader (get fuses to do it) to guess if it's Optiboot at all
2. Check 2 first words of bootloader for 'rjmp' instuction. Both should be 'rjmp'.
If 1. and 2. is true, it's very likely that this is Optiboot with flash write feature.

uzi18, 2017-08-08:
it is impossible to read first 2 bytes.
They are randomized.

majekw, 2017-08-08:
Other bootloaders often have fuses set to prevent reading and writing bootloader part of flash.
Optiboot allows reading bootloader memory, so it's definitely possible to read it.

uzi18, 2017-08-09:
I think now about such case:
1.write into eeprom/rtc marker - "unsupported"
2.try to write known data with do_spm, call bootloader+2 bytes/1word - as
in optiboot.
3.write into eeprom/rtc marker - "supported"

0.on reboot check if marker is setted to unsupport.

bergus, 2017-12-28:
I have used the following code to verify the boot loader contents before jumping into it (calling `do_spm`):
```c
// using do_spm requires majekw's optiboot bootloader
// this function checks whether it is installed and provides the function at the do_spm address
// this requires to lock fuses to be unset - if they're not, we cannot access the bootloader section
// not going to hack around it...  http://oneweekwonder.blogspot.de/2014/07/bootjacker-amazing-avr-bootloader-hack.html
bool pgm_validate_optibootloader() {
  static const __flash uint8_t spm_code[] = {0xFC, 0x01, 0x0A, 0x01, 0x67, 0xBF, 0xE8, 0x95, 0x11, 0x24, 0x07, 0xB6, 0x00, 0xFC, 0xFD, 0xCF, 0x66, 0x70, 0x29, 0xF0, 0x45, 0x2B, 0x19, 0xF4, 0x81, 0xE1, 0x87, 0xBF, 0xE8, 0x95, 0x08, 0x95};
  uint16_t vec = pgm_read_word(2 * (uint16_t) (void*) do_spm);
  if ((vec & 0xF000) != 0xC000) // rjmp
    return false;
  uint16_t boot_code = ((uint16_t) (void*) do_spm + (vec & 0x0FFF) + 1) * 2;
  for (uint16_t i=0; i<sizeof(spm_code); i++)
    if (pgm_read_byte(boot_code + i) != pgm_read_byte(&spm_code[i]))
      return false;
  return true;
}
```
However, one needs to set the lock bits to enable reading data from the boot section, which is not possible with the bootloader but requires SPI programming.