majekw/optiboot #9: Erase without setting RWWSRE bit

bergus, 2017-12-28:
Hello,
I am facing [this issue](http://www.avrfreaks.net/forum/avr-spm-instruction-usage-peculiar-behavior-soliciting-input?page=all) when using `do_spm`. In short:

> Datasheets mention two methods for self-programming flash.
>  I can't get the first one to work:
>> Alternative 1, fill the buffer before a Page Erase
>> * Fill temporary page buffer
>> * Perform a Page Erase
>> * Perform a Page Write
> 
> This approach results in all 0xFF in the page.

As the thread turned out (and restored my sanity) this is not due to a hardware bug, but the undocumented feature (and wrong sample code) that re-enabling the RWW section after the page erase sequence does void the temporary page buffer.

Would it be possible that Optiboot supports this? I see that there is already a conditional, but I do not really understand what it does. Can I somehow use Optiboot to erase, write, and only then reenable rww? If not, could you update the bootloader to support this feature?

majekw, 2017-12-28:
1.  In fact it's documented :-)
`The temporary buffer will auto-erase after a Page Write operation or by writing the RWWSRE bit in
SPMCSR` - RWWSRE is bit to reenable RWW

2. Optiboot already supports this:
- your code calling `do_spm` MUST be in NRWW section, somewhere below bootloader (chip limitation)
- for ERASE you need to pass to `do_spm` data byte which is not equal 0 (so erase wont try to reenable RWW and will not destroy write buffer), see https://github.com/majekw/optiboot/blob/supermaster/optiboot/bootloaders/optiboot/optiboot.c#L1039

bergus, 2017-12-30:
Thanks a lot! I was blind, focusing only on the [weird `_BV` things](http://www.atmel.com/webdoc/avrlibcreferencemanual/FAQ_1faq_use_bv.html) in line 1066 and the comment regarding it. I totally missed the (now obvious) `&& (data == 0)` or the documentation regarding the whole function. I'm sure I can  get this working now.

closed 2017-12-30 by bergus