majekw/optiboot #10: do_spm on atmega2560 not working

ikkentim, 2018-03-21:
Hi,

I flashed the bootloader onto a atmega2560 and tried to use `do_spm` from program space using `optiboot_page_erase`, but it crashes and jump backs to 0x0000. Has anyone ever tried this? Any ideas why this might happen?

majekw, 2018-03-21:
Have you tried _flash_program.ino_ example before?
Restarting is often caused by bootloader without do_spm feaure.
And yes, I tested it on amtega2560 and it worked.

ikkentim, 2018-03-26:
I think I somehow messed up the bootloader, reflashed it and it works fine again. Thanks!

ikkentim, 2018-03-26:
I'm sorry, but I've still got some problems... The demo program flash_program.ino works fine, but with higher addresses it's still broken. If I take the demo flash_program and modify it to use addresses 0x20000 or 0x10000 it breaks:

```
Current flash contents:
r⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..⸮⸮..~⸮..|⸮..z⸮..x⸮..v⸮..t⸮..⸮⸮..p⸮..^⸮..3⸮..j⸮..h⸮..f⸮..d⸮..b⸮..`⸮..^⸮..\⸮..Z⸮..X⸮..V⸮..T⸮..R⸮..P⸮..N⸮..L⸮..J⸮..H⸮..F⸮..D⸮..B⸮..@⸮..>⸮..<⸮..:⸮..8⸮..6⸮..4⸮..2⸮..0⸮..R$⸮⸮⸮⸮⸮޿Ϳ.⸮⸮⸮⸮⸮⸮⸮⸮⸮
Type 256 characters to store in flash:
1234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456
All chars received
Erasing buffer
```
The garbage as current content is, of course, expected. But the program just gives up after "erasing buffer". I've dumped the flash and it seems it has erased page 0 instead of the proper page. Addressses 0x00 to 0x100 are filled with 0xff

ikkentim, 2018-03-26:
I'm guessing the RAMPZ isn't getting used by the SPM?

ikkentim, 2018-03-27:
Test code for reference: https://gist.github.com/ikkentim/2555a24c2a60ee19428aaee8370b4b70

majekw, 2018-03-30:
I see the problem. You are using: ``optiboot_page_erase((optiboot_addr_t)(void*) &((uint8_t *)MEMADDR)[0]);`` where MEMADDR is just a #define.

optiboot_page_* functions expect address in memory as first argument. In example there is ``optiboot_addr_t)(void*) &flash_buffer[0]``, but all that magic with casting etc. is to get address of flash_buffer variable.

So, when you use it with arbitrary address as you do, it's enough just to call: ``optiboot_page_erase(MEMADDR)``

ikkentim, 2018-04-09:
Oh wow! That worked! I wonder what kind of magic is going on here that fails to make it work 🤔

ikkentim, 2018-04-09:
Hmmpf... It fails when I do this:
```
optiboot_addr_t ad = MEMADDR;
optiboot_page_erase(ad);
```

That should work, right? No weird casting going on here...

ikkentim, 2018-04-10:
I've done some more testing and I'm getting more and more confused. After I flash the bootloader the flashing ( https://gist.github.com/ikkentim/4fd26e12e3d1c2ecd582186ca35c1402 ) works only a few times, and then it starts crashing again

ikkentim, 2018-04-10:
I've done even more testing:

- I used the original `flash_program.ino` from this repo
- Without modifications it runs fine for many times in a row
- When changing the `flash_buffer` variable to `const char *flash_buffer = 0x5000;` it runs fine for many times in a row
- When changing the `flash_buffer variable to `const char *flash_buffer = 0x8000;` I can flash just once, after that, the first do_spm call causes a reset of the program.
  - After the first occurance of this bug, even going back to the original `flash_program.ino`
  - To get the original `flash_program.ino` working again, I have to re-flash optiboot.

ikkentim, 2018-04-17:
@majekw Do you have any suggestions for how I can solve this issue?

ikkentim, 2018-04-23:
I modified the do_spm function in the bootloader to handle cli/SREG for me, write 256 bytes to a far flash page and this works without fail. So I don't know what the problem is with, do_spm but calling it repeatedly somehow makes it fail. But calling it just once with all the data it needs make it work just fine.

ikkentim, 2018-05-02:
It seems now that it is the way do_spm is called. 
This is working for me right now:
```
  asm volatile (
    "movw r4, %0\n"
    "movw r2, %1\n"
    "call %2\n"
    :
    : "r" ((uint16_t)addr),
      "r" ((uint16_t)datpt),
      "i" ((uint32_t)((FLASHEND-1023+2)))
    : "r4", "r5", "r2", "r3"
  );
```
I'm reading the registers in the bootloader afterwards using locals bound to these registers.

closed 2018-05-02 by ikkentim