majekw/optiboot #11: Choose to compile without flash write functionality

MCUdude, 2018-08-07:
Hi!
I'm currently working on my [optiboot_flash make script](https://github.com/MCUdude/optiboot_flash/blob/master/makeall). The current version of (the official) optiboot bootloader now works with EEPROM upload too, and IIRC it will compile down to less than 512B with EEPROM write functionality is on.

My fork is (heavily) based on your fork, and I haven't understood low level details of the flash write functionality you implemented.

I'd like to be able to disable your flash writing functionality (as an option in the build script), enable EEPROM support and still end up with a file less than 512B. Is this possible?

majekw, 2018-08-08:
How much memory you want to save? There was only 4 bytes difference between flash and no-flash version.
If you want to get 4B back, you can just remove pre_main() function safely.

But I'm pretty sure that Optiboot could be optimized to get that memory from other parts without sacrificing functionality :-)
Just observe https://github.com/Optiboot/optiboot/pull/142 as I'll try to get this memory from somewhere to fit flash write with eeprom in following days.

MCUdude, 2018-08-08:
OK, 4B is nothing. I thought it might be something like 40B. westfw wrote something about getting optiboot with EEPROM support to fit within 512B. I'd very much like to see this happened!
https://github.com/Optiboot/optiboot/issues/235

majekw, 2018-08-17:
@MCUdude , I made small optimization in size to original Optiboot, but it should work also with Optiboot_flash. It gives 6 bytes of room, not very much, but should be enough to fit also flash functionality into original Optiboot.
Are you able to test it on some real chips? I have only Atmega328 and Atmega2560 in my drawer, so I'll try to test it today/tomorrow, but more is better :-)

Here is a commit: https://github.com/majekw/optiboot/commit/bf652ba9c021345c251386b906a435de9292086d